Flawioo/Creality-Ender-3-board-v4.2.7 #1: Creality v4.2.7

Emile58-F, 2020-08-28:
Hello. Excuse me, this is the right connection, to install a bl touch. Because here I bought from the Amazon.fr site but the hookups with 5 pins is the same creality v4.2.7 motherboard you can help me please ... Good night Emile58-F

Flawioo, 2020-08-29:
> Hello. Excuse me, this is the right connection, to install a bl touch. Because here I bought from the Amazon.fr site but the hookups with 5 pins is the same creality v4.2.7 motherboard you can help me please ... Good night Emile58-F

Hi @Emile58-F sure I'll try to help you. Please can you get some pictures and share here with me? I'm not sure if i got your problem...

Emile58-F, 2020-08-29:
Good evening. the bltouch model is not the same i don't see the color code, i place the photos you requested. I think I was wrong I took for a cr-10 instead of ender 3d, yet I saw which was compatible.

M. Emile Fautres
[image]
[image]

Flawioo, 2020-08-30:
> Good evening. the bltouch model is not the same i don't see the color code, i place the photos you requested. I think I was wrong I took for a cr-10 instead of ender 3d, yet I saw which was compatible.
> 
> M. Emile Fautres
> [image]
> [image]

Hi, Try this:
[image]

It supposed you may have to change the wire order to get it working.
First of all check servo (blue, red and yellow) and only after make them work ok lets to next step that is the probe itself (black and white).

Flawioo, 2020-09-01:
No feedback, I think the question iwas solved.

closed 2020-09-01 by Flawioo