Flawioo/Creality-Ender-3-board-v4.2.7 #2: M600 command not active

Toldwin, 2020-08-31:
Hi, 

I open an issue but you don't have to fix it. It's more to give you feedback, fell free to close the issue without resolving it.
I used your simple firmware binary for my new v4.2.7 motherboard because I found M600 command wasn't working on stock creality firmware. This test shown that your binary (simplest) is also not managing M600 command.

Could be great to add this function... 
This video probably shows modifications that has to be set : https://www.youtube.com/watch?v=6023oBYmXW4
If I find time I ll try to also self build a firmaware but I have a young kid and time is short so not sure I will find time for this.

Anyway thanks much for your work !
Your firmware works well on a stock ender 3 v1 (with only V4.2.7 motherboard upgrade)
Best regards

Flawioo, 2020-09-03:
Hi @Toldwin I've received your feedback.
I've watched the video and I kindly added the M600 command to a new firmware following link:

https://github.com/Flawioo/Creality-Ender-3-board-v4.2.7/raw/master/firmware-20200903-164823-FilChange-m600-enabled.bin

closed 2020-09-03 by Flawioo

Toldwin, 2020-09-05:
Many thanks Flawioo ! 👍 

I will test this ASAP.