Flawioo/Creality-Ender-3-board-v4.2.7 #3: Z-axis won't home

tamorgen, 2020-09-09:
Flawioo,
I followed your steps, and it is not working for me properly on my Ender 3 4.2.7 board.  The BL Touch will not home the Z-Axis.    It will only go up, not down, when trying to home.  I tried both your compiled .bin file, and tried compiling it on my own.

After a week, I finally got Creality to send me a compiled firmware that does work, but it is very stripped down and is running an old version of Marlin.  I'd like to get it working on 2.0.x code.

Any ideas?

Emile58-F, 2020-09-19:
Salut 
Excusez mois, j'ai un petit problème avec la carte mère creality v4.2.7, je ne sais pas si c'est la carte mère qui a un problème où le firmware, parce que ici je me fait un petit objet avec deux tiroirs je fais le boîtier tout aller bien, hier le 18 septembre, je veux faire le tiroir. J'ai l'axe (x) qui se déplace vers la gauche au lieu à droite, j'ai regardé pour voir si aurez pas un fil déconnecté tout est normal. je ne sais pas pourquoi que l'axe (X) fait ça. Tout fonctionnait très bien j'ai même était content de la carte mère, mais que je suis triste. Vous pouvez m'aider s'il vous plaît. Je ne sais si c'est la carte mère ou le firmware Surtout me demande pas de modifier des textes dans le logiciel du firmware je ne serais pas 

pourquoi quel part du mauvais côté. à chaque fois que j'appuis sur Auto Home elle par à droite de mauvais coter. Pour le moment je cherche un firmware pour la mettre à niveau.
A+

Flawioo, 2020-09-23:
> Salut
> Excusez mois, j'ai un petit problème avec la carte mère creality v4.2.7, je ne sais pas si c'est la carte mère qui a un problème où le firmware, parce que ici je me fait un petit objet avec [...]

Hi! Please use this pre compiled firmware: https://github.com/Flawioo/Creality-Ender-3-board-v4.2.7/blob/master/firmware-20200903-164823-Simplest%2BFilChange-m600-enabled.bin
Best Regards

Flawioo, 2020-09-23:
> Flawioo,
> I followed your steps, and it is not working for me properly on my Ender 3 4.2.7 board. The BL Touch will not home the Z-Axis. It will only go up, not down, when trying to home. I tried [...]

Hi @tamorgen sorry for the delay I was on vacation. Let's check your issue.
I think you are using the most complete firmware I compiled.
You have to connect the 3 servo wire from BLTouch to BL_T (on board) and those 2 probe pins (usualy black and white) on Z- on board. If any problem try to swap black and white order.

Please give me a feedback if you can.
Best Regards.

Flawioo, 2020-09-23:
Here is the example how I've connected:
[image]

tamorgen, 2020-09-23:
Flawioo,
No worries, you are allowed to have a vacation!

The way you have it in the picture is the way I am currently configured, and it does work.   I am curious however why it doesn’t work connecting all 5 pins to the BL_T connection when compiling per the directions in your Wiki.

Change: //#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
To: #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN // Enable BLTouch probe pins (white and black wires) to be connected to Z-
or
Change: //#define Z_MIN_PROBE_PIN 32 // Pin 32 is the RAMPS default
To: #define Z_MIN_PROBE_PIN 17 // Creality V4.2.7 BLTouch OUT (written on the board)

I did eventually get Creality to send me a version that works with those ports, so I could get the BL Touch to work for the first time, but it’s a severely crippled (and older) version of Marlin.  Teaching Tech on YouTube published a video for this board suggesting using the dedicated Z-stop port, and I tried that, and it’s worked since (as I initially stated above)

Since Creality sent me a firmware that works on the BL_T connection, it should be possible to compile my own and make it function as intended as well.

Right now, I’m kind of in a "It’s not broke, don’t fix it” position, but if I end up compiling new firmware in the future with updated Marlin, it would be worth trying to get it to work again on the dedicated port.

Emile58-F, 2020-09-23:
Bonsoir.

Je vous en remercie beaucoup à tous de m'avoir aider, mais malheureusement je suis parvenue à réparer ma carte mère Creality, c'était pas à cause du firmware, j'ai remplacer une pièces sur la carte mère. tout est revenue normal elle vas encore mieux qu'avant, encore merci à tous.

Bonne soirée
A+

Flawioo, 2020-10-06:
Without update for long time, I believe the problem was solved.

closed 2020-10-06 by Flawioo