Flawioo/Creality-Ender-3-board-v4.2.7 #4: Runout Sensor on silent board

PaNaVTEC, 2020-09-19:
Hey @Flawioo I was following this guide and this post: https://github.com/MarlinFirmware/Marlin/issues/19029#issuecomment-680974552 of yours, I have everything working except the runout sensor, do you know the wiring of the 3 pin jst connector of the stock 4.2.7 board to connect a single microswitch? I tried to connect the regular `C, NC` pins of the microswitch to different positions of the silent board switch but I cound't make it work. Something like this: 

[image]

Any help is appreciated, thanks :)

PaNaVTEC, 2020-09-19:
Lol, just wiring the 2nd pin as well started to work I thought the two on the sides were the only ones needed, but nop. Anyway, it works, maybe it helps anyone 😁

closed 2020-09-19 by PaNaVTEC

Flawioo, 2020-09-20:
Hi @, You got the point, just two pins are enough. 
Best Regards

ShaunTheBaa, 2021-04-02:
I dont understand, how did you fix it? I have a micro switch wired exactly like the picture above and it doesn't work, M119 just always shows it as triggered. What 2nd pin are you talking about?

k0jara, 2021-04-07:
Switch has 3 pins:
g - ground
c - closed
nc - not closed

by default, the connection ground/closed is closed, ground/not_closed is open
triggering the switch, these states change: gound/closed is open, ground/not_closed is closed

i might have mixed it up, but you get the idea how the switch works.

docwoelle, 2021-06-17:
> I dont understand, how did you fix it? I have a micro switch wired exactly like the picture above and it doesn't work, M119 just always shows it as triggered. What 2nd pin are you talking about?

You have to use a pullup resistor (10k) between V and S. The switch (NO and C) must be between G and S.
In Marlin, set "FIL_RUNOUT_STATE" to "TRIGGERED".

jx2014, 2021-07-25:
The idea is to either send 3.3V to the Signal pin (input logic High), or ground the Signal pin (input logic Low).
Therefore, the Signal pin from the 4.2.7 PCBA must go to 'Common' of the micro switch. 

You never ever want to connect the Ground pin from the 4.2.7 PCBA to the Switch 'Common' pin, doing so will short circuit the 3.3V output from the main board and damage it.

jx2014, 2021-07-25:
This is the default setting in `Configuration.h`

>#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define FIL_RUNOUT_ENABLED_DEFAULT true // Enable the sensor on startup. Override with M412 followed by M500.
  #define NUM_RUNOUT_SENSORS   1          // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.
  #define FIL_RUNOUT_STATE     LOW        // Pin state indicating that filament is NOT present.
  #define FIL_RUNOUT_PULLUP               // Use internal pullup for filament runout pins.
  //#define FIL_RUNOUT_PULLDOWN           // Use internal pulldown for filament runout pins.
  //#define WATCH_ALL_RUNOUT_SENSORS      // Execute runout script on any triggering sensor, not only for the active extruder.
                                          // This is automatically enabled for MIXING_EXTRUDERs.

As you can see, the Run state is by default `Low`, if the switch is `Open`, the mainboard input signal pin should get a `Low` input signal, in other words, the ground pin from the mainboard should be connected to the signal pin by the switch mechanism. 

When the filament is present, the roller switch is depressed, the mainboard input signal pin should get a `High` input signal, in other words, 3.3V pin from the mainboard should be connected to the signal pin by the switch mechanism. However, there is a `#define FIL_RUNOUT_PULLUP`, which means the signal pin, if nothing is connected to it, will remain `High` state due to a internal pull up resistor that's connected to the power supply, in other words, you don't really need to connect the 3.3V pin to the swtich at all.

some terms to describe the three pins of the roller switch: 
NC. Normally Closed, when the roller is relexed, the NC pin is shorted to the Common pin while NO pin becomes open.
NO. Normally Open, when the roller is depressed, the NO pin is shortted to the Common pin while NC pin becomes open.
C.    Common.

jonathanpere404, 2023-06-06:
[image]
[image]
[image]

this is how i did it.

aldi1502, 2023-06-13:
is there a particular place to put it in the program

Flawioo, 2023-06-13:
Hi @aldi1502 sorry but I can't help you with v4.2.2 board.

aldi1502, 2023-06-13:
ok thanks

yw4z, 2024-05-31:
@jonathanpere404 do you have any issues on this method. i'm planning to use this on klipper for klicky probe. but can't find any info. also i got 4.2.7 board that has separate slot for BLTOUCH and i'm not sure can i connect an micro switch to it's socket

ANGELESCALONABUENO, 2025-01-15:
Hello, how are you, I wanted to ask if it is possible to use this part of the filament detector as an end stop for The reason why it tells me that the end stop is activated all the time despite not having anything connected, the truth is that it is a board that I had lying there and it wouldn't seem strange to me if it is damaged, so I thought In two plans A to use the filament sensor and interpret it as end stop of x and B it would be to transfer the x to y and the y to z and buy a blt touch, do you see the first option as viable? The truth is I thought I had made a mistake with the pins but in reality I didn't.

[image]

jonathanpere404, 2025-01-16:
Haven't been here in a while.
@aldi1502 You probably got it done, I guess. Anyway, this video has a good explanation of how to get started with Marlin installation on the board: https://www.youtube.com/watch?v=80IWoygNxB4
That video also explains how to get the BLTouch to work as an X-axis sensor.

@yw4z For me, the filament sensor wasn't the best because when the filament runs out, the end tip of the filament is usually tilted.

[image]

The extruder motor is not powerful enough to pull it through. So the switch is always "ON," and it thinks that there is filament still. I think you're asking about the Z-axis end-point sensor.

[image]

So you want to install the "Klicky Probe" on it. Mine originally came with a "Klicky Probe," but I had some Chinese copy of an Ender 3. Actually, this was my original Z-axis sensor:

[image]

It wasn't good because it has the metal spring thing, just not accurate enough. You can just follow the video but skip the BLTouch changes.

@ANGELESCALONABUENO I think your problem is this: if you use Marlin, there are options to change the signal type. "NO" = Normally Open, or "NC" = Normally Closed. You just change the "false" to "true" what ever axis you choose. Here’s an image of Marlin version 2.1.2.1. There might be a newer version, but i got the old one.

NO (Normally Open):

-In its default state, the circuit is open, meaning the electrical connection is broken, and no current flows.
-When the switch or relay is activated, the circuit closes, allowing current to flow.

NC (Normally Closed):

-In its default state, the circuit is closed, meaning the electrical connection is complete, and current flows.
-When the switch or relay is activated, the circuit opens, stopping the current.

[image]

this is how my original printer looked like and how the z-axel sensor was (Voxelab Aquila C2), but i bought the board 4.2.7 for it. 

[image]

[image]