Flawioo/Creality-Ender-3-board-v4.2.7 #6: V4.2.7 Physical board connections

hankster42, 2020-09-26:
The Bin file you provided, until I learn how to compile my own, was exactly what I've been looking for.  My current project is adding a BLTouch and Runout Filament sensor to my Ender 3 v4.2.7 board.  The question I have is this.  The new V4.2.7 board has a 5pin JST socket designed for the BLTouch.  I plan to install a JST connector on the 5 leads coming from the BLTouch and its' 1.5M extension that I purchased.  Pin 1  G - Brown, Pin 2 V-Red, Pin 3 IN-Yellow, Pin 4-G-Black and Pin 5-White.   The Runout Filament Sensor comes with a 3Pin JST plug which will go into the 3Pin socket next to the X limit switch.  The 3 wires connecting to S G and V are 1-White, 2-Black, and 3-Red.   When filament is present Pin S is set High connected to V.  The when filament is out then S is connected to G or Signal Low.  Does the wiring of these two components as described conform to the Firmware Configuration provided in your BIN file?

Flawioo, 2020-10-03:
Hi! I'm not sure if I'll answer exactly what you are looking for, if not please let me know.

The runout sensor (if not optical) you can use just two wires, it is enough to trigger, use center pin with left pin.
The Black and white wires of your BLTouch have to be connected to -Z, the original end stop switch isn't needed anymore.

Best Regards

closed 2020-10-06 by Flawioo

hankster42, 2020-10-06:
Yes you are correct the issue has been resolved.