Flawioo/Creality-Ender-3-board-v4.2.7 #8: Several redefine warnings when setting #define Z_MIN_PROBE_PIN 17

rmrighes, 2020-10-08:
I see the current bugfix branch from Marlin has _#define Z_MIN_PROBE_PIN PB1_ in _pins_CREALITY_V4.h_, which is included at the end of _pins_CREALITY_V427.h_.

Here's an excerpt of the output when I follow the instructions and set _#define Z_MIN_PROBE_PIN 17_ instead. Which PIN is the correct one to use in the v4.2.7 board while using the 5-pin BL_T connector? 

Compiling .pio\build\STM32F103RET6_creality\src\src\sd\Sd2Card.cpp.o
                 from Marlin\src\module\../inc/MarlinConfig.h:32,
                 from Marlin\src\module\../MarlinCore.h:24,
                 from Marlin\src\module\planner.h:33,
                 from Marlin\src\module\planner.cpp:65:
Marlin\src\module\../inc/../pins/stm32f1/pins_CREALITY_V4.h:84:0: warning: "Z_MIN_PROBE_PIN" redefined
 #define Z_MIN_PROBE_PIN                     PB1   // BLTouch IN
 
In file included from Marlin\src\module\../inc/MarlinConfigPre.h:37:0,
                 from Marlin\src\module\../inc/MarlinConfig.h:28,
                 from Marlin\src\module\../MarlinCore.h:24,
                 from Marlin\src\module\planner.h:33,
                 from Marlin\src\module\planner.cpp:65:
Compiling .pio\build\STM32F103RET6_creality\src\src\sd\SdBaseFile.cpp.o
Marlin\src\module\../inc/../../Configuration.h:860:0: note: this is the location of the previous definition
 #define Z_MIN_PROBE_PIN 17 // Creality V4.2.7 BLTouch OUT (written on the board)

In file included from Marlin\src\module\../inc/../pins/stm32f1/pins_CREALITY_V427.h:48:0,
                 from Marlin\src\module\../inc/../pins/pins.h:569,
                 from Marlin\src\module\../inc/MarlinConfig.h:32,
                 from Marlin\src\module\motion.h:31,
                 from Marlin\src\module\motion.cpp:27:
Marlin\src\module\../inc/../pins/stm32f1/pins_CREALITY_V4.h:84:0: warning: "Z_MIN_PROBE_PIN" redefined
 #define Z_MIN_PROBE_PIN                     PB1   // BLTouch IN

In file included from Marlin\src\module\../inc/MarlinConfigPre.h:37:0,
                 from Marlin\src\module\../inc/MarlinConfig.h:28,
                 from Marlin\src\module\motion.h:31,
                 from Marlin\src\module\motion.cpp:27:
Marlin\src\module\../inc/../../Configuration.h:860:0: note: this is the location of the previous definition
 #define Z_MIN_PROBE_PIN 17 // Creality V4.2.7 BLTouch OUT (written on the board)
 
In file included from Marlin\src\module\../inc/../pins/stm32f1/pins_CREALITY_V427.h:48:0,
                 from Marlin\src\module\../inc/../pins/pins.h:569,
                 from Marlin\src\module\../inc/MarlinConfig.h:32,
                 from Marlin\src\module\probe.cpp:27:
Marlin\src\module\../inc/../pins/stm32f1/pins_CREALITY_V4.h:84:0: warning: "Z_MIN_PROBE_PIN" redefined
 #define Z_MIN_PROBE_PIN                     PB1   // BLTouch IN

In file included from Marlin\src\module\../inc/MarlinConfigPre.h:37:0,
Compiling .pio\build\STM32F103RET6_creality\src\src\sd\SdFatUtil.cpp.o
                 from Marlin\src\module\../inc/MarlinConfig.h:28,
                 from Marlin\src\module\probe.cpp:27:
Marlin\src\module\../inc/../../Configuration.h:860:0: note: this is the location of the previous definition
 #define Z_MIN_PROBE_PIN 17 // Creality V4.2.7 BLTouch OUT (written on the board)

In file included from Marlin\src\module\../inc/../pins/stm32f1/pins_CREALITY_V427.h:48:0,
                 from Marlin\src\module\../inc/../pins/pins.h:569,
                 from Marlin\src\module\../inc/MarlinConfig.h:32,
                 from Marlin\src\module\servo.cpp:27:
Marlin\src\module\../inc/../pins/stm32f1/pins_CREALITY_V4.h:84:0: warning: "Z_MIN_PROBE_PIN" redefined
 #define Z_MIN_PROBE_PIN                     PB1   // BLTouch IN

In file included from Marlin\src\module\../inc/MarlinConfigPre.h:37:0,
                 from Marlin\src\module\../inc/MarlinConfig.h:28,
                 from Marlin\src\module\servo.cpp:27:
Marlin\src\module\../inc/../../Configuration.h:860:0: note: this is the location of the previous definition
 #define Z_MIN_PROBE_PIN 17 // Creality V4.2.7 BLTouch OUT (written on the board)

In file included from Marlin\src\module\../inc/../pins/stm32f1/pins_CREALITY_V427.h:48:0,
                 from Marlin\src\module\../inc/../pins/pins.h:569,
                 from Marlin\src\module\../inc/MarlinConfig.h:32,
                 from Marlin\src\module\settings.h:28,
                 from Marlin\src\module\settings.cpp:46:
Marlin\src\module\../inc/../pins/stm32f1/pins_CREALITY_V4.h:84:0: warning: "Z_MIN_PROBE_PIN" redefined
 #define Z_MIN_PROBE_PIN                     PB1   // BLTouch IN
[... 150 lines cut ...]

Compiling .pio\build\STM32F103RET6_creality\liba29\U8glib-HAL\clib\u8g_com_arduino_no_en_parallel.c.o
In file included from Marlin\src\sd\../inc/MarlinConfigPre.h:37:0,
                 from Marlin\src\sd\../inc/MarlinConfig.h:28,
                 from Marlin\src\sd\SdFile.cpp:30:
Compiling .pio\build\STM32F103RET6_creality\liba29\U8glib-HAL\clib\u8g_com_arduino_parallel.c.o
Compiling .pio\build\STM32F103RET6_creality\liba29\U8glib-HAL\clib\u8g_com_arduino_port_d_wr.c.o
Marlin\src\sd\../inc/../../Configuration.h:860:0: note: this is the location of the previous definition
 #define Z_MIN_PROBE_PIN 17 // Creality V4.2.7 BLTouch OUT (written on the board)
Compiling .pio\build\STM32F103RET6_creality\liba29\U8glib-HAL\clib\u8g_com_arduino_ssd_i2c.c.o

Compiling .pio\build\STM32F103RET6_creality\liba29\U8glib-HAL\clib\u8g_com_arduino_st7920_custom.c.o
In file included from Marlin\src\sd\../inc/../pins/stm32f1/pins_CREALITY_V427.h:48:0,
                 from Marlin\src\sd\../inc/../pins/pins.h:569,
                 from Marlin\src\sd\../inc/MarlinConfig.h:32,
                 from Marlin\src\sd\cardreader.cpp:23:
Marlin\src\sd\../inc/../pins/stm32f1/pins_CREALITY_V4.h:84:0: warning: "Z_MIN_PROBE_PIN" redefined
 #define Z_MIN_PROBE_PIN                     PB1   // BLTouch IN

In file included from Marlin\src\sd\../inc/MarlinConfigPre.h:37:0,
                 from Marlin\src\sd\../inc/MarlinConfig.h:28,
                 from Marlin\src\sd\cardreader.cpp:23:
Marlin\src\sd\../inc/../../Configuration.h:860:0: note: this is the location of the previous definition
Compiling .pio\build\STM32F103RET6_creality\liba29\U8glib-HAL\clib\u8g_com_arduino_st7920_hw_spi.c.o
 #define Z_MIN_PROBE_PIN 17 // Creality V4.2.7 BLTouch OUT (written on the board)

rmrighes, 2020-10-08:
Compiling the firmware with #define Z_MIN_PROBE_PIN PB1 does not generate any of those warnings.

Flawioo, 2020-10-08:
> Compiling the firmware with #define Z_MIN_PROBE_PIN PB1 does not generate any of those warnings.

Hi!
You are right, I'm working on the latest Marlin version that received the correct line parameters and can give you a better experience, please have a try on this:
https://github.com/Flawioo/Creality-Ender-3-board-v4.2.7/tree/Marlin-2.0.7

Any feedback will be appreciated.
BR

closed 2020-10-08 by Flawioo