Flawioo/Creality-Ender-3-board-v4.2.7 #9: Just to make sure

zevv, 2020-11-11:
Hey,

I recently upgraded my board to the v4.2.7 for reasons of noise reduction, but I lost my M600

Just to make sure: firmware-20200903-164823-Simplest+FilChange-m600-enabled.bin is basically similar to the stock Creality v0.0.6, but with M600 enabled?

btw, does this board support upgrades by putting the firmware on SD or do I need to go through USB?

Thanks!

Flawioo, 2020-11-13:
> stock Creality v0.0.6

Hi! Sorry I can't confirm it to you because I've never used stock firmware from creality because them never have worked to me.

closed 2020-11-13 by Flawioo

zevv, 2020-11-13:
Aah right. Well, I tried yours on my Ender3, but the endstop inputs seem to mismatch the firmware, so homing is broken to start with.

Thanks anyway!

Flawioo, 2020-11-13:
> Aah right. Well, I tried yours on my Ender3, but the endstop inputs seem to mismatch the firmware, so homing is broken to start with.

Ohh, seriously? Nobody have said it to me before. If I got a new info like that I'll make a troubleshooting to get that. You are welcome.

Flawioo, 2020-11-13:
I'll  really appreciate, thank you. Good prints!