Flawioo/Creality-Ender-3-board-v4.2.7 #10: Before install...

Asaiel65, 2020-11-29:
Hi, and thx for all the work :-), seems excellent.

I have a few question before trying your compiled firmware. I have an Ender 3 (V1) with 4.2.7 mainboard, no BL-Touch, No End filament detector.
What is the version of FW to download? Will I recover M600?
If any pbm, is it still possible to roll back to stock FW?

Thx

Flawioo, 2020-12-01:
Hi! The version you have to download is the second one with the filament change option and if you note any problem you can roll back to stock FW no problem. It's super easy to do it. Just put the firmware in a SDCard, insert and reboot the printer. After few seconds the firmware will be installed. The same processes to roll back.

closed 2020-12-01 by Flawioo

Asaiel65, 2020-12-03:
Thx a lot, almost worked « out of the box » 😁

Just three more questions:
- There is no printer name displayed on the LCD. Not really an issue, I was just wondering...
- The M600 command works: printer pauses, head moves, filament unloads and wait for load, goes through purge, and then immediatly resumes the print, when I expected it would wait for user ton confirm resume by clicking or would offer possibility to purge more... is there any way I can fix it?
- Is there any way to get the sources for this FW, so as to only modify a few things?

Regards,

Flawioo, 2020-12-03:
There is no printer name displayed on the LCD. Not really an issue, I was just wondering...
The M600 command works: printer pauses, head moves, filament unloads and wait for load, goes through purge, and then immediatly resumes the print, when I expected it would wait for user ton confirm resume by clicking or would offer possibility to purge more... is there any way I can fix it?
Is there any way to get the sources for this FW, so as to only modify a few things?

The printer name is just because the original sources came without them, but I'll adjust in the next compilation proccess.
That length cannot be configured from the printer menu but it is the ADVANCED PAUSE PURGE LENGTH in the configuration section that you can modify in the compilation proccess. I'll try to adjust a bit more to the next firmware version.
Sure you always can get the sources for this firmware, here you go:
Download bugfix from git hub https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip
Download Configuration exemples: https://github.com/MarlinFirmware/Configurations/archive/bugfix-2.0.x.zip

Good prints!