Flawioo/Creality-Ender-3-board-v4.2.7 #11: Motors are not moving

EranIDar, 2021-01-14:
Hi,

I first did my own firmware and decided to try your Simple Firmware but my motors are not moving anymore...

I can't do : 
- AutoHome
- MoveAxis

Event tho the motors are under tension...

Can you please help me please ? 

Thank you in advance,

fix218, 2021-01-14:
> Hi,
> 
> I first did my own firmware and decided to try your Simple Firmware but my motors are not moving anymore...
> 
> I can't do :
> 
> * AutoHome
> * MoveAxis
> 
> Event tho the motors are under tension...
> 
> Can you please help me please ?
> 
> Thank you in advance,

Are you using a BL touch on the default port on the board?
Do you having any accessories plugged in like filament detection or anything like that?

EranIDar, 2021-01-14:
Hi,

No I don't use a BL touch or any accessories...

fix218, 2021-01-14:
Which firmware version did you use?

fix218, 2021-01-14:
I am using this one with no problems.
[https://github.com/Flawioo/Creality-Ender-3-board-v4.2.7/tree/Marlin-2.0.7](url)

EranIDar, 2021-01-14:
Hi,

I'm using Marlin bugfix-2.0.x

Maybe I should try Marlin 2.0.x instead

fix218, 2021-01-14:
The link I just posted I have tested it and it works perfectly fine. I'm
currently trying to get 2.0.7.2 compiled now.

fix218, 2021-01-14:
> I'm using Marlin bugfix-2.0.x
> 
> Maybe I should try Marlin 2.0.x instead

 I would try the 2.0.7.1 that I listed in the link that he has. give it a try first, like I said it's been working great for me.

EranIDar, 2021-01-14:
I tried marlin 2.0.x and ender 3 Pro V1.5 for the config and it works !

Thank you for your help !

EranIDar, 2021-01-14:
Actually it doesn't work haha..

I launched a print and bed get stuck at 45 °C even if the command is 55 °C. The bed is increasing to 45 and get stuck there...

Weird

EranIDar, 2021-01-14:
And the same, if I put 65 °C as a command I get 55 °C on the bed

fix218, 2021-01-14:
That sucks.. I just uploaded a newer version of the firmware I was just
testing it so far everything worked great You're welcome to try it. Let me
know if you have any problems.

fix218, 2021-01-14:
Sounds like you're maximum values are set too low

EranIDar, 2021-01-14:
I'll try again...
I've watch the settings in configuration.h and the minimum value for bed is 0 and max is 120.

fix218, 2021-01-14:
Check your configuration for the board type I know I've seen in the
configuration file v1.5 it actually has the wrong board identified which
will cause problems

fix218, 2021-01-14:
I've been using the ender 3 Pro 4.2.7 configuration files not the v1.5

EranIDar, 2021-01-15:
OK it seems to work.

I managed to make it work using : 

- Marlin bugfix 2.0.x
- Configuration : Creality V427

But I had to invert every DIRECTION and STEP pins in Marlin > src > pins > stm32f1 > pins_creality_V427.h
This file has been modified 23 days ago and I think the error might come from it

I'll try a xyz cube today

Thanks for your help

closed 2021-01-27 by EranIDar